iralabdisco/p3dx-driver-board #2: Funzionamento TD340 (Motor Driver)

giuseppe-caliaro, 2025-03-28:
Il circuito integrato ST TD340 permette il pilotaggio di un motore DC attraverso un ponte-H.
Di seguito il diagramma a blocchi e la descrizione del pinout.

**_Diagramma a blocchi/Descrizione pinout TD390_**
[image]

**Punti chiave appresi:**

- L1/L2: Pilotano i gate dei mosfet low-side del ponte-H, Q1L e Q2L rispettivamente in figura.

- H1/H2: Pilotano i gate dei mosfet high-side del ponte-H, Q1H e Q2H rispettivamente in figura.

- S1/S2: Costituiscono insieme a CB1 e CB2 il circuito di bootstrap del ponte H.
Il circuito di boostrap è necessario per superare la tensione di soglia (Vgate - Vsource) del mosfet high-side quando quest'ultimo è ON. Perchè accade? questo accade perchè in questa configurazione Source del mosfet è a 12V.
Diversamente nei mosfet low-side ciò non accade, essendo source a GND quando i mosfet low-side sono attivi. vedi https://www.homemade-circuits.com/h-bridge-bootstrapping

- IN1/IN2: Questi pin sono utilizzati per controllare velocità (IN1) e direzione (IN2) del motore. La modalità "freno" è attivata quando ci sono 0V sul pin IN1 per piu di 200 microsecondi.

**Questioni aperte:** 

1. Nel circuito sono presenti due Op-Amp U6A/U6B (fisicamente uno solo, Op-Amp TS9221 presenta due ingressi invertenti, due ingressi non invertenti e due output). Non mi è completamente chiaro il loro utilizzo.
U6A sembra essere configurato come comparatore (vedi partitore di tensione su ingresso invertente)
U6B avendo una retroazione positiva (vedi Vout applicata a ingresso non invertente) sembra essere un comparatore con isteresi (Trigger di Schmitt).
L'uscita dell'Op-Amp U6B controlla la base di un BJT, il quale agisce come interruttore per i segnali RIGHT_C1 e RIGHT_C2 provenienti dalla MCU.

Non mi è chiaro il motivo del collegamento del Drain di Q5 con il circuito sopra citato.
Non trovo riscontro di continuità con questo collegamento _DrainQ5 - R36 - Ingresso non invertente U6A_ ,  mentre trovo riscontro di continuità con _SourceQ7 - R36 - GND_ , il mosfet Q7 costituisce parte del low-side del ponte H quindi è sensato che il Source sia a GND.

[image]

Per un controllo di base del TD340, IN1 e IN2 sono sufficienti, tuttavia, la comprensione del circuito sopra descritto penso sia fondamentale per decifrare la logica di controllo dei segnali MCU verso il TD340.

jack23247, 2025-03-28:
Ottimo lavoro.

Ti lascio qui una risorsa interessante sui ponti H: https://www.modularcircuits.com/blog/articles/h-bridge-secrets/

Per quanto riguarda Q5, ci guardiamo insieme quando posso. Gli opamp anche, ma non sono molto ferrato.. quattro occhi sono sempre meglio di due però

giuseppe-caliaro, 2025-03-28:
Credo di aver capito che: 
Gli ingressi degli operazionali sono collegati a Source di Q7 e Q8 (low-side), attraverso questi mosfet scorre la corrente che scorre attraverso il motore.
Interposta tra questo collegamento è posizionata una resistenza di shunt (R2 nello schema), la quale viene normalmente utilizzata come misuratore di corrente (qui una discussione circa la misurazione di corrente attraverso l'utilizzo di una resistenza di shunt https://electronics.stackexchange.com/questions/152500/measure-current-through-h-bridge-using-shunt).

La caduta di tensione ai capi della resistenza di shunt è proporzionale alla corrente che scorre attraverso i mosfet. 
Se si dovesse verificare un overcurrent, causerebbe una caduta di tensione, probabilmente sotto una certa soglia dimensionata nell'implementazione degli op-amp. Questa caduta di tensione pilota il BJT, bloccandone la conduzione e interrompendo cosi i segnali RIGHT_C1 e RIGHT_C2.

Sostanzialmente mi sembra di aver capito che è un circuito di protezione da overcurrent.

giuseppe-caliaro, 2025-03-28:
Condivido il nuovo schema (Right motor driver) aggiornato con le correzioni riguardanti gli op-amp discussi precedentemente:

[image]

giuseppe-caliaro, 2025-03-29:
**Ulteriori note interessanti circa il funzionamento del driver TD340:**

- Frequenza di switching: 25kHz

- Lavora con una tensione tra 6.5V e 18.5V

- Da datasheet: _"The speed control (or duty cycle) is achieved by the Low Side Drivers which impose the PWM function while the cross-corresponding High Side MOSFETS is kept fully ON."_.
Questa tecnica consiste nell'applicare il segnale PWM ai mosfet low-side, cioè quelli connessi a massa, mentre i mosfet high-side, cioè quelli connessi alla tensione di alimentazione, vengono mantenuti sempre ON.

- Ci sono 4 pin che forniscono servizi aggiuntivi:

1. **Reset**: TD340 fornisce un circuito di supervisore di tensione, il quale riavvia il microcontrollore quando la tensione di quest'ultimo scende sotto una soglia di 4.2V, e l'uscita Reset resta a LOW fino a quando la tensione non supera i 4.3V, dopodichè il microcontrollore riprende la sua normale esecuzione. 

1. **Watchdog** (WD nello schema): TD340 fornisce un circuito di watchdog, il quale resetta il driver per evitare che il sistema rimanga bloccato in caso di errori software/hardware. Watchdog controlla che il microcontrollore invii segnali costanti al pin WD, se questi impulsi non avvengono per un periodo troppo lungo (timeout, settabile attraverso il dimensionamento di un condensatore), questo blocca il sistema finchè il microcontrollore non riprende ad inviare impulsi al pin WD.

1. **Temp**: TD340 fornisce un indicatore di temperatura.

1. **Standby** (STBY nello schema): TD340 può essere messo in standby mode tramite software, i mosfet vengono spenti. Questo permette di ridurre il consumo a 200 microAmpere. Per uscire da questa modalità il sistema ha bisogno di 20 ms per caricare il condensatore facente parte del circuito di boostrap. Il driver non deve ricevere segnali IN1 e IN2 prima che il condensatore sia completamente carico, altrimenti i mosfet high-side potrebbero non funzionare correttamente.

[image]

Da schema i sopracitati pin ad eccezione di Standby risultano però essere stati esclusi nel design della driver-board.
La configurazione di Standby come accennato non risulta essere esclusa, tuttavia dal connettore del microcontrollore non sembra essere presente un pin designato a questa funzione:

[image]

Questa questione è cruciale per la sicurezza e l'affidabilità complessiva del sistema, poiché, con l'esclusione dei pin relativi a Reset e Watchdog nel design della driver-board, la gestione delle sicurezza e affidabilità è completamente delegata al microcontrollore STM32F767ZI (MCU scelta per pilotare la driver-board).
Credo quindi che sarà necessario tenere conto di questo aspetto in fase di sviluppo.

jack23247, 2025-03-31:
Ottimo lavoro. Alcune osservazioni:

- Bello e leggibile lo schema. Al posto del motore però metterei il connettore, dato che il motore non è tecnicamente parte della scheda. Possiamo comunque mettere il motore in un box per chiarezza.
- Avevo notato anche io il fatto che non è stato usato il circuito di Watchdog interno ai TD340. Non so il perché di questa scelta, ma sicuramente ci sarà stato un motivo ingegneristicamente valido. Comunque teniamone conto.

Per quanto riguarda i "salti", quando passo in lab ci guardiamo insieme!

giuseppe-caliaro, 2025-03-31:
Perfetto, allora modifico lo schema sostituitendo il motore con il connettore.

Per la questione watchdog magari studiando l'interfacciamento tra la nostra board e la driver-board salta fuori il motivo dell'esclusione di questi pin.

giuseppe-caliaro, 2025-03-31:
Il modello è: **Renesas SH2-7144 RISC 32-bit**. 
Documentazione: https://www.renesas.com/en/document/mah/sh7144-group-sh7145-group-hardware-manual?r=1054851

Che succede se imposti il duty cycle del PWM a zero? E come passi, in generale, il segnale PWM al driver?

Impostando il duty cycle del PWM a zero mi aspetto che il Driver setti a 0V il motore.

In generale, penso che il microprocessore della Nucleo generi il segnale PWM (attraverso un timer).
La Nucleo sarà collegata, attraverso il connettore specifico, alla driver-board, la quale invia il segnale PWM al pin del Driver.

jack23247, 2025-03-31:
Ottimo! Allora possiamo gestirlo così anche noi, secondo me.

giuseppe-caliaro, 2025-04-01:
> * STBY è attivo basso (ha la linea sopra), quindi se non erro R41 fa da resistenza di pull-up e fondamentalmente esclude la funzionalità di Stand-by.

Grazie per la precisazione

closed 2025-05-22 by giuseppe-caliaro