iralabdisco/p3dx-driver-board #5: User Control Panel

giuseppe-caliaro, 2025-04-04:
User Control Panel è un pannello di controllo posizionato a lato del Pioneer.
Vi sono presenti:

- 4 Pulsanti:
1. **AUX1**
2. **AUX2**
1. **RESET**
1. **MOTORS**

- 7 Led:
1. Due led relativi alla trasmissione e ricezione della porta seriale
2. Due led relativi ai pulsanti AUX1 e AUX2
3. Un led relativo allo stato della batteria
4. Un led di stato (STAT)
5. Un led di PWR, probabilmente un led di stato dei motori (ON/OFF)

Inoltre sono presenti un buzzer e una porta seriale.

---

**Pannello Pioneer P3DX:**
[image]

**PCB pannello**
[image]

**Pinout connettore MCU-Pannello (Lato MCU)**
[image]

Al fine di utilizzare nel modo corretto gli elementi presenti sul pannello, è necessario capirne il funzionamento.

Sulla scheda è presenta un circuito integrato, il 74HC00, il quale è  costituito da 4 NAND, le quali condividono lo stesso ingresso (quindi parliamo sostanzialmente di NOT).

Di seguito il pinout del 74HC00 da datasheet:

[image]

Non mi però è del tutto chiaro il funzionamento dell'integrato, soprattutto per quanto riguarda la gestione degli ingressi e delle uscite:

[image]

giuseppe-caliaro, 2025-04-04:
Condivido lo schema relativo alla **porta seriale** presente sulla scheda del pannello:

[image]

giuseppe-caliaro, 2025-04-04:
Condivido lo schema relativo al **buzzer** presente sulla scheda del pannello:

[image]

giuseppe-caliaro, 2025-04-04:
Condivido lo schema relativo ai **pulsanti** e relativi **led** presenti sulla scheda del pannello:

[image]

giuseppe-caliaro, 2025-04-04:
Rimane dunque da chiarire la questione relativa all'integrato 74HC00 e i relativi pin IN/OUT, i quali probabilmente gestiscono il led relativo alla stato della batteria

giuseppe-caliaro, 2025-04-05:
Ulteriori note interessanti riguardo il funzionamento dello User Control Panel (ARCOS based, quindi MCU e Firmware proprietario):
Da Manuale Operativo:

-**Led batteria:**
_Pag. 14:_
[image]
_Pag. 11:_
[image]

---

-**Led PWR, Led STAT:**
_Pag. 10:_
[image]

---

-**Buzzer:**
_Pag. 11:_
[image]

---

-**Serial Port:**
_Pag. 11:_
[image]

---

-**Power Switches (AUX1/AUX2 buttons):**
_Pag. 11:_
[image]

---

-**Reset/Motors buttons**
_Pag. 11/12:
[image]
[image]

---

Concludo dicendo che secondo me sarebbe utile chiarire cosa vogliamo effettivamente replicare del pannello originale, soprattutto per quanto riguarda i due pulsanti degli AUX power switches: potremmo considerarli, come suggerito da te in laboratorio,  semplicemente come pulsanti utente, da usare liberamente per altre funzioni, non per forza legati all’alimentazione.

giuseppe-caliaro, 2025-04-05:
Questo è quello che sono riuscito a capire riguardante al circuito del led relativo alla batteria e il circuito integrato 74HC00:

Il diodo led è bicolor, perchè a differenza dei led RGB a tre pin, il led in questione ne presenta due.
Quando la corrente scorre attraverso il diodo in un verso, emette luce verde, quando la corrente scorre nel verso opposto, emette luce rossa. (vedi: https://www.quarktwin.com/blogs/led/bi-color-led-working-principle/21?srsltid=AfmBOop2j-tdDkJ70AU99Gy2rPILpXY14Xsb4I_RKeu6brBk5deMyBv-)

L'integrato 74HC00 viene utilizzato per creare una differenza di potenziale e permette alla corrente di scorrere in un verso o nel verso opposto.

[image]

Quando PLED (che arriva dal microcontrollore, il quale è processato secondo la tensione della batteria) ha valore HIGH, l'uscita di U1B è 0V, mentre l'uscita di U1A è 5V, questo permette alla corrente di scorrere in un verso.
Viceversa quando PLED ha valore LOW, l'uscita di U1B è 5V, mentre l'uscita U1A è 0V, questo permette alla corrente di scorrere nel verso opposto.
 
Da Manuale Operativo:
[image]
L'arancione credo che venga ottenuto con un segnale PWM generato dal microntrollore, il quale commutando ad alta frequenza il verso della corrente, genera il colore arancione/giallo.

jack23247, 2025-04-07:
Ottimo lavoro. L'unica cosa che cambierei è il simbolo usato per i pulsanti e quello usato per i toggle switch (oppure aggiungerei una nota).

Io sinceramente lascerei AUX1 e AUX2 programmabili dall'utente. L'interfaccia seriale a 3 pin potremmo sia usarla che scartarla e collegare direttamente l'UART alla scheda. Idealmente direi di mettere dei jumper per permettere agli utenti di fare quello che vogliono.

Molto interessante il LED bicolore per il controllo del voltaggio batteria... anche se mi sembra un po' over-engineered. Possiamo pilotarlo dal micro oppure prendere un integrato che faccia da comparatore/voltmetro. Oppure collegare direttamente un voltmetro a 7 segmenti e non usare quel LED, che forse è anche meglio.

giuseppe-caliaro, 2025-04-07:
Grazie per i suggerimenti, provvedo quindi a modificare lo schema.

Riguardo ad AUX1 e AUX2, potremmo discutere di quale funzionalità assegnare loro in base alle nostre esigenze.

L'idea del voltmetro con display a 7 segmenti mi sembra ottima, possiamo in tal modo monitorare il valore esatto di tensione delle batterie.

closed 2025-05-22 by giuseppe-caliaro