iralabdisco/p3dx-driver-board #7: Integrazione Nucleo↔Driver-Board

giuseppe-caliaro, 2025-04-15:
In questa issue discuterò l'integrazione della Nucleo stm32 F767ZI con la driver-board del Pioneer 3DX, inizalmente dal punto di vista elettrico/alimentazione, e succcesivamente dal punto di vista della gestione dei segnali (PWM per controllo motori, lettura encoder, pulsanti ecc.)

- **Alimentazione**
Da User manual STM32 Nucleo-144 boards Pag. 22:

[image]

Come descritto la Nucleo può essere alimentata in tre modi differenti, Vin, E5V e 3.3V, tutti con un loro range di tensioni tollerate e un limite di assorbimento di corrente.

Essendo la driver-board del Pioneer alimentata a 5V per la gestione della circuiteria di controllo dei motori e per l'interfacciamento con la MCU proprietaria, e 12V per alimentare i due motori DC, credo che sia corretto supporre che utilizzeremo la modalità E5V per alimentare la Nucleo. 
Questo implica come specificato dal manuale nel settare corretamente il jumper di JP3 per selezionare l'alimentazione corretta, e rimuovere il jumer di JP1 per eliminare il limite a 100mA di assorbimento di corrente.

Come mostrato nella figura sottostante la driver-board fornisce i 5V per alimentare la MCU attraverso il connettore J13, mentre lato MCU proprietaria il connettore è collegato a J22
[image]

Il pinout del connettore J13 è il seguente:

[image]
Di questi pin solo i pin 1,2,3 raggiungono l'MCU proprietaria. 
Oltre ai 5V, il connettore J13 fornisce 12V , i quali sono utilizzati dallo User Control Panel, attraverso il connettore J11 (come si vede sopra in figura), per l'array di Sonar e vari componenti ausiliari come Joystick e giroscopio.

Fammi sapere se l'analisi relativa all'alimentazione della Nucleo è secondo te corretta.

giuseppe-caliaro, 2025-04-15:
Condivido inoltre una sequenza definita **_mandatory_**:

[image]

Spiega quali siano i passaggi per alimentare esternamente la Nucleo e avere comunque la possibilità di flashare o debuggare attraverso l'USB di debug, evitando di riscontrare i problemi descritti nell'immagine sopra.

giuseppe-caliaro, 2025-04-15:
Intendi la questione jumper JP3? In tal caso sì ho fatto qualche esperimento alimentando la Nucleo esternamente ed è molto semplice

giuseppe-caliaro, 2025-04-16:
**- PWM e direzione motori**
Al fine di controllare i motori del Pioneer, come abbiamo discusso nella issue dedicata all'IC TD340, la Nucleo deve generare 4 segnali:
1. PWM motore sinistro
2. PWM motore destro
3. Segnale HIGH/LOW per direzione motore sinistro
4. Segnale HIGH/LOW per direzione motore destro

La famiglia STM32F7, alla quale appartiene la nostra Nucleo, fornisce diversi timer per la gestione del segnale PWM, in particolare fornisce quattro timer General purpose: TIM3 e TIM4 a 16 bit, e TIM2 e TIM5 a 32 bit, come mostrato in figura;
[image]

Da datasheet dell'IC TD340, la frequenza massima con la quale è in grado di commutare il segnale è di 25KhZ.
Credo quindi che possiamo optare per l'utilizzo dei timer TIM3 e TIM4 a 16 bit e ottenere comunque una risoluzione più che sufficiente in termini di sottolivelli. (Esempio potrei impostare ad 1 il prescaler, quindi utilizzare senza divisori direttamente la  frequenza massima del clock del sistema ottenendo addirittura 3839 sottolivelli, quindi penso che 16 bit siano più che sufficienti per un controllo preciso dei motori).

Posso quindi impostare il pin GPIO **PA6** come TIM3_CH1 (PWM Generation Channel 1 attraverso il timer TIM3) e **PD12** come TIM4_CH1 (PWM Generation Channel 1 attraverso il timer TIM4).
Questi due segnali permettono quindi di controllare la **velocità del motore** di destra e del motore di sinistra del Pioneer.

---

Per quanto riguarda la gestione della **direzione dei motori**, l'IC TD340 richiede come detto prima un segnale per motore.
Il valore di questo segnale deve essere HIGH o LOW e determina il senso di rotazione del motore.
A questo scopo penso che possiamo utilizzare due pin GPIO qualsiasi, magari tenendo conto semplicemente di non utilizzare dei pin GPIO che possono implementare Alternate Function particolari, così da evitare di utilizzare inutilmente un pin GPIO con funzionalità avanzate per un semplice toggle.

---

Fammi sapere se questa analisi ti sembra corretta così procedo oltre.

giuseppe-caliaro, 2025-04-17:
Intanto confermo che da datasheet PE9 e PE11, e PA0 e PA1 sono 5V tolerant, nella colonna _I/O Structure_ **FT** significa 5V tolerant:

**_PE9/PE11_**
[image]

**_PA1_**
[image]

**_PA0_**
[image]

jack23247, 2025-04-18:
- Ottima analisi. Per i pin puoi guardare il firmware scritto da Federica per Otto che trovi sul GitHub di IRALab per vedere che pin ha usato lei all'epoca 
- Per passare a 3v3 a segnali TTL possiamo usare un level-shifter che ci fa anche da buffer

jack23247, 2025-04-18:
Per gli input non lo so. Usare un level-shifter è problematico perché potrebbe distorcere il segnale che arriva dagli encoder in maniera troppo significativa.

giuseppe-caliaro, 2025-04-22:
Per quanto riguarda l'incompatibilità dei segnali output di **direzione** e **PWM** con la driver-board del Pioneer potremmo utilizzare un **Logic Level Shifter** come quello mostrato in figura:

[image]

giuseppe-caliaro, 2025-04-22:
Analizzando il percorso che i segnali seguono dal connettore _Motors_ presente sulla MCU proprietaria noto che:
- I segnali di **PWM** e **Direzione** provenienti dal processore prima di arrivare al connettore ed essere trasmessi alla driver-board passano attraverso ad un **HCT541**.
HCT541 è un buffer non invertente, il quale ha il compito di ripetere i segnali in uscita, amplificandone la corrente e isolando unidirezionalmente la MCU (quindi anche proteggendola) dalla driver board.
- I segnali di **Encoding** (lo stesso vale per i segnali provenienti dai sensori di prossimità) provengono invece dal connettore e prima di arrivare al processore passano attraverso un **LVC541A**.
L' LCC541A, come l'HCT541 è un buffer, con la differenza che opera con tensioni di uscita minori. Questo perchè i progettisti della MCU-board del Pioneer hanno dovuto interfacciarsi con due tensioni differenti (come noi), quella dei segnali di encoding a 5V e quella dei pin del processore a 3.3V (difatti sono presenti due regolatori di tensione **LT1086** sulla MCU-board, uno per alimentare il processore, l'altro per alimentare i due integrati **LVC541A**, uno per i segnali di encoding l'altro per i sensori di prossimità).

Penso quindi che i due integrati sopra descritti, l'**HCT541** e il **LVC541A** possano essere una soluzione concreta anche per il nostro problema di interfacciamento.

Cosa ne pensi? I due integrati, alimentati rispettivamente a 5V l'integrato che gestisce direzione e PWM e 3v3 l'integrato che gestisce i segnali di encoding, sono quindi dei Level Shifter che possiamo utilizzare?

giuseppe-caliaro, 2025-04-23:
Assumendo che il ragionamento relativo agli integrati che ho descritto precedentemente sia corretto, ho fatto una prima bozza di come collegherei la Nucleo e la driver-board attraverso la board di interfaccia:

[image]
Qualche considerazione:
- Devo assicurarmi di cortocircuitare a GND i segnali di Enable degli integrati HCT541 e LVC541A al fine di attivarne le uscite.
- LVC541A assorbe corrente nell'ordine delle decine di mA, quindi penso che la Nucleo riesca a gestirle. Possiamo comunque con un regolatore di tensione abbassare i 5V provenienti dalla driver-board a 3v3 e alimentare direttamente l'integrato, evitando così di sovraccaricare inutilmente la Nucleo. Lo schema in questione è infatti quello consigliato da datasheet:
[image]

- Da datasheet viene raccomandato di mettere a GND tutti i pin non utilizzati:
[image]
- Utilizzare un bypass capacitor per il pin di Vcc per evitare dei disturbi dell'alimentazione: _Each VCC pin should have a good bypass capacitor to prevent power disturbance. For devices with a single supply, 0.1 μf is recommended;_

giuseppe-caliaro, 2025-04-30:
Per quanto riguarda la questione latenza introdotta dai buffer, da datasheet l'IC HCT541 introduce un ritardo _tipico_ di 12 ns, mentre l'IC LVC541A introduce un ritardo _massimo_ di 5.1 ns.

jack23247, 2025-05-02:
Ok, come vedi l'integrato migliore è stato usato per gli encoder perché sono più sensibili alla latenza. Controlla anche la tensione max che possono fornire i due IC

giuseppe-caliaro, 2025-05-09:
Come discusso in laboratorio, è sensato "mappare" tutti i segnali che provengono da J11 e J28 al fine di poterli utilizzare per implementazioni future, ad esempio per la gestione dei sensori oppure per la gestione dei pin analogici, senza modificare radicalmente l'Interface-Board o crearne una nuova.

A tale scopo serve ridefinire lo schema condiviso in precedenza, sia per gestire i segnale della porta seriale (come discusso nella issue di integrazione dello User Control Panel) sia per gestire tutti quei segnali che da J11 a J28 che al momento non sono gestiti dalla prima bozza di schema della Interface-Board.

- **Sensori IR**:
Dal connettore J28 abbiamo otto pin relativi ai sensori IR, uno per ogni sensore presente sul Pioneer. Dato che questi segnali devono essere adattati ad un livello di tensione corretto per la Nucleo, necessitano di un LVC541A, il quale ha esattamente otto ingressi. 
Detto questo ciascuo degli otto bit relativi ai sensori devono essere gestiti tramite un pin GPIO in modalità Input.
Stm32 permette di leggere con una sola istruzione il registro di un'intera porta GPIO, dal registro IDR, quindi sarebbe sensato utilizzare un'intera porta GPIO per catturare questi dati, cosi da averne un accesso più rapido per gli sviluppi futuri.

- **Analog Input:**
Il connettore J28 trasporta due pin per degli ingressi analogici, i quali possono essere ad esempio i dati da un un giroscopio oppure un joystick. Questi segnali devono essere processati lato Nucleo da due convertitori analogico digitali. 
La MCU Board proprietaria non gestisce direttamente i segnali dal connettore agli ADC, ma probabilmente passano attraverso una circuiteria necessaria ad adattare il livello di tensione. Al momento non ho ancora trovato riscontro di continuità di questi pin, quindi devo approfondire meglio la questione.

- **Sensore di temperatura:**
Sulla Driver-Board è presente un sensore di temperatura LM35DT, il quale fornisce la temperatura dei mosfet del ponte H. 
Questo sensore tramite una circuiteria composta da amplificatori comparatori controlla automaticamente una ventola tramite il connettore J37.
Controllando la documentazione del Pioneer, la temperatura viene direttamente fornita dal connettore J38, il quale non è stato utilizzato dalla MCU Board proprietaria e non è trasmesso da nessun connettore, difatti nella documentazione è etichettato come _unused_.
[image]
Potremmo, tramite il convertitore analogico-digitale presenta sulla Nucleo processare questo dato, utilizzando i dati forniti dal datasheet del sensore di temperatura LM35DT. 
L'LM35DT fornisce un valore di tensione proporzionale alla temperatura, "_Linear + 10-mV/°C Scale Factor_" da datasheet, quindi fornirebbe un valore di tensione piuttosto piccolo e gestibile dalla Nucleo, però in caso di guasto potrebbe danneggiare l'ADC.
Ritengo quindi sia necessario, nel caso in cui volessimo utilizzare il sensore di temperatura, progettare una circuiteria che protegga l'ADC in caso di guasti e/o sovratensioni.

- **AUX1 e AUX2:**
I segnali di AUX1 e AUX2 provenienti dal J11 sono gestiti come pulsanti input dalla Nucleo. Gli stessi segnali, ma sul connettore J28 sono utilizzati per attivare o disattivare la circuiteria di alimentazione ausiliaria presente sulla Driver-Board, descritta dal seguente schema:
[image]
Tra i pin AUX1 e AUX2 del connettore J11 e AUX1 e AUX2 del connettore c'è difatti continuità, questo significa quindi che quando sullo User Control Panel viene premuto il pulsante di AUX1 e/o il pulsante di AUX2, la circuiteria di alimentazione sopra descritta viene attivata. 
Dobbiamo quindi gestire questa situazione, perchè da un lato vogliamo che i pulsanti AUX1 e AUX2 siano utilizzati come pulsanti utente, dall'altro lato la Driver-Board è stata progettata per attivare o disattivare parti della sua circuiteria direttamente da questi pulsanti. Dobbiamo discutere di come gestire questa situazione.

- **Motor Enable:**
Il segnale di Motor Enable presente sul connettore J28 è un segnale di output del processore, sulla MCU Board passa attraverso l'HCT541 per adattarlo al livello di tensione della Driver-Board.
Lato Driver-Board, ho riscontrato delle difficoltà nel trovare la funzionalità di questo segnale.
Chiaramente il nome è auto-esplicativo e abilita i motori, tuttavia non trovo un riscontro pratico del segnale ad esempio sul ponte-H TD340 o in un altra posizione della Driver-Board. In lab insieme possiamo provare e risolvere la questione.
Possiamo gestire questo segnale semplicemente utilizzando un qualsiasi pin GPIO impostato come output, e adattandolo attraverso l'utilizzo di un HCT541.

- **E-Stop Input:** 
Il segnale di E-Stop è stato descritto approfonditamente, è un segnale inviato in input al processore con la funzione di notificare che il pulsante di E-Stop, collegato alla Driver-Board attraverso l'opportuno connettore, è stato premuto.
Come abbiamo discusso in laboratorio noi vogliamo un pulsante di E-Stop che bypassi il software e abbia un effetto direttamente sull'hardware, ad esempio collegandolo in serie alle batterie. 
La nostra Driver-Board (vedi immagine sotto) al momento non è predisposta per ospitare il pulsante di E-Stop nel connettore che i progettisti hanno designato a questa funzione, tuttavia essendo presente la predisposizione, potrebbe essere valido comunque gestire questo segnale sulla Interface-Board, cosi da poter essere implementato in futuro.
[image]

Detto questo, nei prossimi giorni condivido il nuovo schema con la gestione totale dei segnali dei connettori J28 e J11.

giuseppe-caliaro, 2025-05-10:
Studiando lato Driver-Board il segnale **Motor Enable** del connettore J28, noto che viene elaborato tramite un'ic, l'HC14 cioè un Trigger di Schmitt invertente (mostrato in figura).
[image]
Il segnale di Motor Enable arriva al connettore J28 attraverso le uscite di questo integrato, 1Y e 2Y. Testando l'integrato con il multimetro noto che gli ingressi 1A e 2A sono a massa. Questo mi fa pensare che in normali condizioni questo segnale è di Input alla MCU-Board, il quale avverte il processore di attivare o meno il segnale di PWM e direzione per attivare i motori.
Applicando un segnale alto agli ingressi 1A e 2A, le uscite dell'integrato sarebbero a LOW e di conseguenza anche il segnale di Motor  Enable, permettendo quindi al processore di avere un flag di quando ci siano o meno le condizioni di utilizzare i motori.

Purtroppo non riesco a capire sulla Driver-Board dove e come questo segnale viene commutato, potremmo guardare insieme in laboratorio.
In ogni caso, come detto prima, dobbiamo trattare questo segnale come segnale di Input della Nucleo, quindi dovrà essere adattato attraverso l'utilizzo dell'LVC541A.